Digital-EDA/Digital-IDE #2: [0.3.0 beta]"显示当前文件的netlist"无法正确显示出网表

windmilltown, 2023-07-17:
模块代码
`module clkdiv(
    input clk50,
    input rst_n,
    output reg clkout
);
    reg [15:0] cnt;
    always @(posedge clk50 or negedge rst_n)
    begin
        if(!rst_n)
        begin
            cnt <= 16'b0;
            clkout <= 1'b0;
        end
        else if(cnt == 16'd162) 
        begin
            clkout <= 1'b1;
            cnt <= cnt + 16'd1;
        end
            else if(cnt == 16'd325) 
        begin
            clkout <= 1'b0;
            cnt <= 16'd0;
        end
            else 
        begin
            cnt <= cnt + 16'd1;
        end
    end
endmodule`

显示netlist
[image]

LSTM-Kirigaya, 2023-07-21:
抱歉，我无法复现你的错误，因为我可以可视化出这个网表。

[image]

请问你可以重新尝试或者点击”输出“，选择Yosys来查看输出信息吗？

DuBirdFly, 2023-07-22:
遇到了同样的右侧没有显示的问题, Yosys输出如下:

```
[Info - 01:30:00] type: message, -- Running command `read_verilog -sv -formal -overwrite /d:/PrjWorkspace/rv32/user/src/core/aa.v' --
[Info - 01:30:00] type: message, 1. Executing Verilog-2005 frontend: /d:/PrjWorkspace/rv32/user/src/core/aa.v
[Info - 01:30:00] type: message, Parsing formal SystemVerilog input from `/d:/PrjWorkspace/rv32/user/src/core/aa.v' to AST representation.
[Info - 01:30:00] type: message, Generating RTLIL representation for module `\clkdiv'.
[Info - 01:30:00] type: message, Successfully finished Verilog frontend.
[Info - 01:30:01] type: message, -- Running command `proc' --
[Info - 01:30:01] type: message, 2. Executing PROC pass (convert processes to netlists).
[Info - 01:30:01] type: message, 2.1. Executing PROC_CLEAN pass (remove empty switches from decision trees).
[Info - 01:30:01] type: message, Cleaned up 0 empty switches.
[Info - 01:30:01] type: message, 2.2. Executing PROC_RMDEAD pass (remove dead branches from decision trees).
[Info - 01:30:01] type: message, Marked 3 switch rules as full_case in process $proc$/d:/PrjWorkspace/rv32/user/src/core/aa.v:7$1 in module clkdiv.
[Info - 01:30:01] type: message, Removed a total of 0 dead cases.
[Info - 01:30:01] type: message, 2.3. Executing PROC_PRUNE pass (remove redundant assignments in processes).
[Info - 01:30:01] type: message, Removed 1 redundant assignment.
[Info - 01:30:01] type: message, Promoted 0 assignments to connections.
[Info - 01:30:01] type: message, 2.4. Executing PROC_INIT pass (extract init attributes).
[Info - 01:30:01] type: message, 2.5. Executing PROC_ARST pass (detect async resets in processes).
[Info - 01:30:01] type: message, Found async reset \rst_n in `\clkdiv.$proc$/d:/PrjWorkspace/rv32/user/src/core/aa.v:7$1'.
[Info - 01:30:01] type: message, 2.6. Executing PROC_ROM pass (convert switches to ROMs).
[Info - 01:30:01] type: message, Converted 0 switches.
[Info - 01:30:01] type: message, <suppressed ~2 debug messages>
[Info - 01:30:01] type: message, 2.7. Executing PROC_MUX pass (convert decision trees to multiplexers).
[Info - 01:30:01] type: message, Creating decoders for process `\clkdiv.$proc$/d:/PrjWorkspace/rv32/user/src/core/aa.v:7$1'.
[Info - 01:30:01] type: message,      1/2: $0\cnt[15:0]
[Info - 01:30:01] type: message,      2/2: $0\clkout[0:0]
[Info - 01:30:01] type: message, 2.8. Executing PROC_DLATCH pass (convert process syncs to latches).
[Info - 01:30:01] type: message, 2.9. Executing PROC_DFF pass (convert process syncs to FFs).
[Info - 01:30:01] type: message, Creating register for signal `\clkdiv.\clkout' using process `\clkdiv.$proc$/d:/PrjWorkspace/rv32/user/src/core/aa.v:7$1'.
[Info - 01:30:01] type: message,   created $adff cell `$procdff$19' with positive edge clock and negative level reset.
[Info - 01:30:01] type: message, Creating register for signal `\clkdiv.\cnt' using process `\clkdiv.$proc$/d:/PrjWorkspace/rv32/user/src/core/aa.v:7$1'.
[Info - 01:30:01] type: message,   created $adff cell `$procdff$20' with positive edge clock and negative level reset.
[Info - 01:30:01] type: message, 2.10. Executing PROC_MEMWR pass (convert process memory writes to cells).
[Info - 01:30:01] type: message, 2.11. Executing PROC_CLEAN pass (remove empty switches from decision trees).
[Info - 01:30:01] type: message, Found and cleaned up 2 empty switches in `\clkdiv.$proc$/d:/PrjWorkspace/rv32/user/src/core/aa.v:7$1'.
[Info - 01:30:01] type: message, Removing empty process `clkdiv.$proc$/d:/PrjWorkspace/rv32/user/src/core/aa.v:7$1'.
[Info - 01:30:01] type: message, Cleaned up 2 empty switches.
[Info - 01:30:01] type: message, 2.12. Executing OPT_EXPR pass (perform const folding).
[Info - 01:30:01] type: message, Optimizing module clkdiv.
[Info - 01:30:01] type: message, -- Running command `write_json  /output' --
[Info - 01:30:01] type: message, 3. Executing JSON backend.

```

windmilltown, 2023-07-25:
```
[Info - 17:02:52] type: message, -- Running command `read_verilog -sv -formal -overwrite /d:/study/robei/robei_verilog/robei_verilog/user/src/uart/clkdiv.v' --
[Info - 17:02:52] type: message, 1. Executing Verilog-2005 frontend: /d:/study/robei/robei_verilog/robei_verilog/user/src/uart/clkdiv.v
[Info - 17:02:52] type: message, Parsing formal SystemVerilog input from `/d:/study/robei/robei_verilog/robei_verilog/user/src/uart/clkdiv.v' to AST representation.
[Info - 17:02:52] type: message, Generating RTLIL representation for module `\clkdiv'.
[Info - 17:02:52] type: message, Successfully finished Verilog frontend.
[Info - 17:02:57] type: message, -- Running command `proc' --
[Info - 17:02:57] type: message, 2. Executing PROC pass (convert processes to netlists).
[Info - 17:02:57] type: message, 2.1. Executing PROC_CLEAN pass (remove empty switches from decision trees).
[Info - 17:02:57] type: message, Cleaned up 0 empty switches.
[Info - 17:02:57] type: message, 2.2. Executing PROC_RMDEAD pass (remove dead branches from decision trees).
[Info - 17:02:57] type: message, Marked 3 switch rules as full_case in process $proc$/d:/study/robei/robei_verilog/robei_verilog/user/src/uart/clkdiv.v:11$1 in module clkdiv.
[Info - 17:02:57] type: message, Removed a total of 0 dead cases.
[Info - 17:02:57] type: message, 2.3. Executing PROC_PRUNE pass (remove redundant assignments in processes).
[Info - 17:02:57] type: message, Removed 1 redundant assignment.
[Info - 17:02:57] type: message, Promoted 0 assignments to connections.
[Info - 17:02:57] type: message, 2.4. Executing PROC_INIT pass (extract init attributes).
[Info - 17:02:57] type: message, 2.5. Executing PROC_ARST pass (detect async resets in processes).
[Info - 17:02:57] type: message, Found async reset \rst_n in `\clkdiv.$proc$/d:/study/robei/robei_verilog/robei_verilog/user/src/uart/clkdiv.v:11$1'.
[Info - 17:02:57] type: message, 2.6. Executing PROC_ROM pass (convert switches to ROMs).
[Info - 17:02:57] type: message, Converted 0 switches.
[Info - 17:02:57] type: message, <suppressed ~2 debug messages>
[Info - 17:02:57] type: message, 2.7. Executing PROC_MUX pass (convert decision trees to multiplexers).
[Info - 17:02:57] type: message, Creating decoders for process `\clkdiv.$proc$/d:/study/robei/robei_verilog/robei_verilog/user/src/uart/clkdiv.v:11$1'.
[Info - 17:02:57] type: message,      1/2: $0\cnt[15:0]
[Info - 17:02:57] type: message,      2/2: $0\clkout[0:0]
[Info - 17:02:57] type: message, 2.8. Executing PROC_DLATCH pass (convert process syncs to latches).
[Info - 17:02:57] type: message, 2.9. Executing PROC_DFF pass (convert process syncs to FFs).
[Info - 17:02:57] type: message, Creating register for signal `\clkdiv.\clkout' using process `\clkdiv.$proc$/d:/study/robei/robei_verilog/robei_verilog/user/src/uart/clkdiv.v:11$1'.
[Info - 17:02:57] type: message,   created $adff cell `$procdff$19' with positive edge clock and negative level reset.
[Info - 17:02:57] type: message, Creating register for signal `\clkdiv.\cnt' using process `\clkdiv.$proc$/d:/study/robei/robei_verilog/robei_verilog/user/src/uart/clkdiv.v:11$1'.
[Info - 17:02:57] type: message,   created $adff cell `$procdff$20' with positive edge clock and negative level reset.
[Info - 17:02:57] type: message, 2.10. Executing PROC_MEMWR pass (convert process memory writes to cells).
[Info - 17:02:57] type: message, 2.11. Executing PROC_CLEAN pass (remove empty switches from decision trees).
[Info - 17:02:57] type: message, Found and cleaned up 2 empty switches in `\clkdiv.$proc$/d:/study/robei/robei_verilog/robei_verilog/user/src/uart/clkdiv.v:11$1'.
[Info - 17:02:57] type: message, Removing empty process `clkdiv.$proc$/d:/study/robei/robei_verilog/robei_verilog/user/src/uart/clkdiv.v:11$1'.
[Info - 17:02:57] type: message, Cleaned up 2 empty switches.
[Info - 17:02:57] type: message, 2.12. Executing OPT_EXPR pass (perform const folding).
[Info - 17:02:57] type: message, Optimizing module clkdiv.
[Info - 17:02:57] type: message, -- Running command `write_json  /output' --
[Info - 17:02:57] type: message, 3. Executing JSON backend.
[Info - 17:02:59] type: message, -- Running command `read_verilog -sv -formal -overwrite /d:/study/robei/robei_verilog/robei_verilog/user/src/uart/clkdiv.v' --
[Info - 17:02:59] type: message, 1. Executing Verilog-2005 frontend: /d:/study/robei/robei_verilog/robei_verilog/user/src/uart/clkdiv.v
[Info - 17:02:59] type: message, Parsing formal SystemVerilog input from `/d:/study/robei/robei_verilog/robei_verilog/user/src/uart/clkdiv.v' to AST representation.
[Info - 17:02:59] type: message, Generating RTLIL representation for module `\clkdiv'.
[Info - 17:02:59] type: message, Successfully finished Verilog frontend.
[Info - 17:03:00] type: message, -- Running command `proc' --
[Info - 17:03:00] type: message, 2. Executing PROC pass (convert processes to netlists).
[Info - 17:03:00] type: message, 2.1. Executing PROC_CLEAN pass (remove empty switches from decision trees).
[Info - 17:03:00] type: message, Cleaned up 0 empty switches.
[Info - 17:03:00] type: message, 2.2. Executing PROC_RMDEAD pass (remove dead branches from decision trees).
[Info - 17:03:00] type: message, Marked 3 switch rules as full_case in process $proc$/d:/study/robei/robei_verilog/robei_verilog/user/src/uart/clkdiv.v:11$1 in module clkdiv.
[Info - 17:03:00] type: message, Removed a total of 0 dead cases.
[... cut ...]
[Info - 17:03:00] type: message, Found and cleaned up 2 empty switches in `\clkdiv.$proc$/d:/study/robei/robei_verilog/robei_verilog/user/src/uart/clkdiv.v:11$1'.
[Info - 17:03:00] type: message, Removing empty process `clkdiv.$proc$/d:/study/robei/robei_verilog/robei_verilog/user/src/uart/clkdiv.v:11$1'.
[Info - 17:03:00] type: message, Cleaned up 2 empty switches.
[Info - 17:03:00] type: message, 2.12. Executing OPT_EXPR pass (perform const folding).
[Info - 17:03:00] type: message, Optimizing module clkdiv.
[Info - 17:03:00] type: message, -- Running command `write_json  /output' --
[Info - 17:03:00] type: message, 3. Executing JSON backend.
```

DreamLand7707, 2023-07-29:
同样遇到了无法输出的问题，可能是环境原因？

#### 代码：
```
module des_sample (
    input            clk,
    input            rst,
    output reg [7:0] nums
);

    always @(posedge clk or negedge rst) begin
        if (!rst) begin
            nums = 8'h00;
        end else begin
            nums = nums + 1;
        end
    end

endmodule
```

#### yosys 输出
```
[Info - 17:12:16] type: message, -- Running command `read_verilog -sv -formal -overwrite /d:/Project/VSCode/verilog/usart/user/src/des_sample.v' --
[Info - 17:12:16] type: message, 1. Executing Verilog-2005 frontend: /d:/Project/VSCode/verilog/usart/user/src/des_sample.v
[Info - 17:12:16] type: message, Parsing formal SystemVerilog input from `/d:/Project/VSCode/verilog/usart/user/src/des_sample.v' to AST representation.
[Info - 17:12:16] type: message, Generating RTLIL representation for module `\des_sample'.
[Info - 17:12:16] type: message, Successfully finished Verilog frontend.
[Info - 17:12:16] type: message, -- Running command `proc' --
[Info - 17:12:16] type: message, 2. Executing PROC pass (convert processes to netlists).
[Info - 17:12:16] type: message, 2.1. Executing PROC_CLEAN pass (remove empty switches from decision trees).
[Info - 17:12:16] type: message, Cleaned up 0 empty switches.
[Info - 17:12:16] type: message, 2.2. Executing PROC_RMDEAD pass (remove dead branches from decision trees).
[Info - 17:12:16] type: message, Marked 1 switch rules as full_case in process $proc$/d:/Project/VSCode/verilog/usart/user/src/des_sample.v:7$1 in module des_sample.
[Info - 17:12:16] type: message, Removed a total of 0 dead cases.
[Info - 17:12:16] type: message, 2.3. Executing PROC_PRUNE pass (remove redundant assignments in processes).
[Info - 17:12:16] type: message, Removed 0 redundant assignments.
[Info - 17:12:16] type: message, Promoted 1 assignment to connection.
[Info - 17:12:16] type: message, 2.4. Executing PROC_INIT pass (extract init attributes).
[Info - 17:12:16] type: message, 2.5. Executing PROC_ARST pass (detect async resets in processes).
[Info - 17:12:16] type: message, Found async reset \rst in `\des_sample.$proc$/d:/Project/VSCode/verilog/usart/user/src/des_sample.v:7$1'.
[Info - 17:12:16] type: message, 2.6. Executing PROC_ROM pass (convert switches to ROMs).
[Info - 17:12:16] type: message, Converted 0 switches.
[Info - 17:12:16] type: message, 2.7. Executing PROC_MUX pass (convert decision trees to multiplexers).
[Info - 17:12:16] type: message, Creating decoders for process `\des_sample.$proc$/d:/Project/VSCode/verilog/usart/user/src/des_sample.v:7$1'.
[Info - 17:12:16] type: message,      1/1: $1\nums[7:0]
[Info - 17:12:16] type: message, 2.8. Executing PROC_DLATCH pass (convert process syncs to latches).
[Info - 17:12:16] type: message, 2.9. Executing PROC_DFF pass (convert process syncs to FFs).
[Info - 17:12:16] type: message, Creating register for signal `\des_sample.\nums' using process `\des_sample.$proc$/d:/Project/VSCode/verilog/usart/user/src/des_sample.v:7$1'.
[Info - 17:12:16] type: message,   created $adff cell `$procdff$4' with positive edge clock and negative level reset.
[Info - 17:12:16] type: message, 2.10. Executing PROC_MEMWR pass (convert process memory writes to cells).
[Info - 17:12:16] type: message, 2.11. Executing PROC_CLEAN pass (remove empty switches from decision trees).
[Info - 17:12:16] type: message, Removing empty process `des_sample.$proc$/d:/Project/VSCode/verilog/usart/user/src/des_sample.v:7$1'.
[Info - 17:12:16] type: message, Cleaned up 0 empty switches.
[Info - 17:12:16] type: message, 2.12. Executing OPT_EXPR pass (perform const folding).
[Info - 17:12:16] type: message, Optimizing module des_sample.
[Info - 17:12:16] type: message, -- Running command `write_json  /output' --
[Info - 17:12:16] type: message, 3. Executing JSON backend.
```

#### 截图：
[image]

DreamLand7707, 2023-07-29:
另外补充一个小细节，就是这个页面的边框拖拽在我这里似乎是有一些问题的。
我想要把黄色的一边拖拽的大一点的话，我会先按住右下角的黑色拖拽点，然后移动鼠标。
但是当我移动鼠标的时候，一旦移动，在移动的那一瞬间黄色一侧就会陡然变窄，并且在拖动过程中，黄色一侧的宽度的增长速度和我的鼠标移动速度并不一致。导致我最终即使把鼠标移动至屏幕最右端时，黄色一侧的大小仍然相较于拖拽前没有改变。

DreamLand7707, 2023-07-29:
> 同样遇到了无法输出的问题，可能是环境原因？
> 
> #### 代码：
> ```verilog
> module des_sample (
>     input            clk,
>     input            rst,
>     output reg [7:0] nums
> );
> 
>     always @(posedge clk or negedge rst) begin
>         if (!rst) begin
>             nums = 8'h00;
>         end else begin
>             nums = nums + 1;
>         end
>     end
> 
> endmodule
> ```
> 
> #### yosys 输出
> ```
> [Info - 17:12:16] type: message, -- Running command `read_verilog -sv -formal -overwrite /d:/Project/VSCode/verilog/usart/user/src/des_sample.v' --
> [Info - 17:12:16] type: message, 1. Executing Verilog-2005 frontend: /d:/Project/VSCode/verilog/usart/user/src/des_sample.v
> [Info - 17:12:16] type: message, Parsing formal SystemVerilog input from `/d:/Project/VSCode/verilog/usart/user/src/des_sample.v' to AST representation.
> [Info - 17:12:16] type: message, Generating RTLIL representation for module `\des_sample'.
> [Info - 17:12:16] type: message, Successfully finished Verilog frontend.
> [Info - 17:12:16] type: message, -- Running command `proc' --
> [Info - 17:12:16] type: message, 2. Executing PROC pass (convert processes to netlists).
> [Info - 17:12:16] type: message, 2.1. Executing PROC_CLEAN pass (remove empty switches from decision trees).
> [Info - 17:12:16] type: message, Cleaned up 0 empty switches.
> [Info - 17:12:16] type: message, 2.2. Executing PROC_RMDEAD pass (remove dead branches from decision trees).
> [Info - 17:12:16] type: message, Marked 1 switch rules as full_case in process $proc$/d:/Project/VSCode/verilog/usart/user/src/des_sample.v:7$1 in module des_sample.
> [Info - 17:12:16] type: message, Removed a total of 0 dead cases.
> [Info - 17:12:16] type: message, 2.3. Executing PROC_PRUNE pass (remove redundant assignments in processes).
> [Info - 17:12:16] type: message, Removed 0 redundant assignments.
> [Info - 17:12:16] type: message, Promoted 1 assignment to connection.
> [Info - 17:12:16] type: message, 2.4. Executing PROC_INIT pass (extract init attributes).
> [Info - 17:12:16] type: message, 2.5. Executing PROC_ARST pass (detect async resets in processes).
> [Info - 17:12:16] type: message, Found async reset \rst in `\des_sample.$proc$/d:/Project/VSCode/verilog/usart/user/src/des_sample.v:7$1'.
> [Info - 17:12:16] type: message, 2.6. Executing PROC_ROM pass (convert switches to ROMs).
> [Info - 17:12:16] type: message, Converted 0 switches.
> [Info - 17:12:16] type: message, 2.7. Executing PROC_MUX pass (convert decision trees to multiplexers).
> [Info - 17:12:16] type: message, Creating decoders for process `\des_sample.$proc$/d:/Project/VSCode/verilog/usart/user/src/des_sample.v:7$1'.
> [Info - 17:12:16] type: message,      1/1: $1\nums[7:0]
> [Info - 17:12:16] type: message, 2.8. Executing PROC_DLATCH pass (convert process syncs to latches).
> [Info - 17:12:16] type: message, 2.9. Executing PROC_DFF pass (convert process syncs to FFs).
> [Info - 17:12:16] type: message, Creating register for signal `\des_sample.\nums' using process `\des_sample.$proc$/d:/Project/VSCode/verilog/usart/user/src/des_sample.v:7$1'.
> [Info - 17:12:16] type: message,   created $adff cell `$procdff$4' with positive edge clock and negative level reset.
> [Info - 17:12:16] type: message, 2.10. Executing PROC_MEMWR pass (convert process memory writes to cells).
> [Info - 17:12:16] type: message, 2.11. Executing PROC_CLEAN pass (remove empty switches from decision trees).
> [Info - 17:12:16] type: message, Removing empty process `des_sample.$proc$/d:/Project/VSCode/verilog/usart/user/src/des_sample.v:7$1'.
> [Info - 17:12:16] type: message, Cleaned up 0 empty switches.
> [Info - 17:12:16] type: message, 2.12. Executing OPT_EXPR pass (perform const folding).
> [Info - 17:12:16] type: message, Optimizing module des_sample.
> [Info - 17:12:16] type: message, -- Running command `write_json  /output' --
> [Info - 17:12:16] type: message, 3. Executing JSON backend.
> ```
> 
> #### 截图：
> [image]

这一问题同样存在于2.6版本

LSTM-Kirigaya, 2023-08-13:
问题解决了，我的vscode开发环境太老了。。。