Digital-EDA/Digital-IDE #3: 例化中的[xx:xx]连线会导致后续颜色错误

DuBirdFly, 2023-07-22:
示例代码:
```verilog
`include "defines.v"

module DataCatch (
    input                               clk,

    input       [3:0]                   rden,
    input       [3:0]                   wren,
    input       [31:0]                  wrdata,
    input       [`DataCatchDepth - 1:0] addr,       // [11:0]

    output wire [31:0]                  rddata

);

ramGen #(
    .Width      ( 8             ),
    .Depth      ( `DataCatchDepth - 2 )
) u0_ramGen (
    .clk        ( clk           ),
    .wren       ( wren[0]       ),
    .wraddr     ( addr[`DataCatchDepth - 1 : 2] ),
    .wrdata     ( wrdata[7:0]   ),
    .rden       ( rden[0]       ),
    .rdaddr     ( addr          ),
    .rddata     ( rddata[7:0]   )
);

ramGen #(
    .Width      ( 8             ),
    .Depth      ( `DataCatchDepth - 2 )
) u1_ramGen (
    .clk        ( clk           ),
    .wren       ( wren          ),
    .wraddr     ( addr          ),
    .wrdata     ( wrdata[7:0]   ),
    .rden       ( rden          ),
    .rdaddr     ( addr          ),
    .rddata     ( rddata        )
);

ramGen #(
    .Width      ( 8             ),
    .Depth      ( `DataCatchDepth - 2 )
) u2_ramGen (
    .clk        ( clk           ),
    .wren       ( wren[2]       ),
    .wraddr     ( addr          ),
    .wrdata     ( wrdata        ),
    .rden       ( rden          ),
    .rdaddr     ( addr          ),
    .rddata     ( rddata        )
);

ramGen #(
    .Width      ( 8             ),
    .Depth      ( `DataCatchDepth - 2 )
) u3_ramGen (
    .clk        ( clk           ),
    .wren       ( wren[3]       ),
    .wraddr     ( addr[`DataCatchDepth - 1 : 2] ),
    .wrdata     ( wrdata[31:24]   ),
    .rden       ( rden[3]       ),
    .rdaddr     ( addr          ),
    .rddata     ( rddata[31:24]   )
);

endmodule
```

[image]

DuBirdFly, 2023-07-24:
另一个颜色时效问题: 标识符(reg, wire)后如果只有1个空格, 则会导致颜色错误

示例代码:
```verilog
module aa(
    input wire clk,
    input      rst,
    output reg ooo
);

reg a;
reg  aa;
reg   aaa;

wire [7:0] b;
wire  [7:0] bb;
wire   [7:0] bbb;

endmodule
```

[image]

DuBirdFly, 2023-07-24:
颜色失效问题3:

[image]

LSTM-Kirigaya, 2023-08-11:
谢谢反馈，应该是正则表达式写炸了，正在排查

LSTM-Kirigaya, 2023-08-11:
修复完成了
[image]

[image]

closed 2023-08-11 by LSTM-Kirigaya