Digital-EDA/Digital-IDE #4: 文件跳转功能失效

qgzln, 2023-07-23:
1文件跳转功能不能使用，左边的
[image]
这些文件都可以出现，并点击进入，但是在top文件里不能通过跳转文件进入到这些文件里
2 点击
[image]
z
会报错：
[image]

qgzln, 2023-07-23:
点击文件跳转的话，在输出那里会显示：
[image]

LSTM-Kirigaya, 2023-08-11:
尝试更新vscode版本后，可以解决 :D

closed 2023-08-11 by LSTM-Kirigaya