Digital-EDA/Digital-IDE #5: [0.3.0 beta] 由于插件不能分析出使用了"include"语法, 从而导致Sim失败

DuBirdFly, 2023-07-24:
Digital-IDE: TreeView显示的工程结构:

[image]

仿真后, Digital-IDE的报错输出:

[image]

iverilog可以用 '-I' 指定 includedir, 要不再整一个 "set as includedir"  😀(*^_^*)

LSTM-Kirigaya, 2023-08-11:
TreeView只显示module之间的层级关系，并不会显示文件。文件只是module的载体，和module的层次关系不一定存在直接关系。

对于您的问题，我们计划在property.json中设计一个新的参数：
```json

"compile" : {
    "include": [
        "<defines.v所在的目录的绝对路径或者相对于运行目录的相对路径>"
    ]
}
```

DuBirdFly, 2023-08-13:
"iverilogCompileOptions" 可以解决 iverilog 的 include 问题, 但我发现 iverilog 和 yosys 对 include 的路径语法不一致:
1. iverilog认为 include 后面只需要跟文件名
2. yosys认为 include 后面需要跟相对路径

[image]

LSTM-Kirigaya, 2023-08-13:
其实iverilog后面也可以根相对路径，但是这个相对路径的父级目录的绝对路径要加入-I参数里面。iverilog整体反正就是比较狗，规则混乱，难以把控，如果你完全不写include，并且保证整个工程没有模块重名，其实是没有问题的。当然，如果你有更好的建议，欢迎提出。

closed 2023-08-15 by LSTM-Kirigaya