Digital-EDA/Digital-IDE #6: [0.3.0 beta] "对当前文件进行仿真"功能报错

DreamLand7707, 2023-07-29:
### 报错截图
[image]

### 报错功能
[image]

### 报错代码
```verilog
module des_sample (
    input            clk,
    input            rst,
    output reg [7:0] nums
);

    always @(posedge clk or negedge rst) begin
        if (!rst) begin
            nums = 8'h00;
        end else begin
            nums = nums + 1;
        end
    end

endmodule
```

### 补充

我找到了正确的解决办法，就是把所有的-y选项移动到要仿真的文件之前，这是iverilog的命令规定
[image]
选项中sourcefile出现在-y之后

只需要把命令更改为如下格式即可：

#### 正确格式：
```
iverilog -g2012 -o "d:/Project/VSCode/verilog/usart/prj/simulation/icarus/out.vvp" -s des_sample -y D:\Xilinx\Vivado\2023.1\data\verilog\src/xeclib -y D:\Xilinx\Vivado\2023.1\data\verilog\src/unisims -y D:\Xilinx\Vivado\2023.1\data\verilog\src/unimacro -y D:\Xilinx\Vivado\2023.1\data\verilog\src/unifast -y D:\Xilinx\Vivado\2023.1\data\verilog\src/retarget "d:/Project/VSCode/verilog/usart/user/src/des_sample.v" D:\Xilinx\Vivado\2023.1\data\verilog\src/glbl.v
```
#### 错误格式：
```
iverilog -g2012 -o "d:/Project/VSCode/verilog/usart/prj/simulation/icarus/out.vvp" -s des_sample "d:/Project/VSCode/verilog/usart/user/src/des_sample.v"   D:\Xilinx\Vivado\2023.1\data\verilog\src/glbl.v -y D:\Xilinx\Vivado\2023.1\data\verilog\src/xeclib-y D:\Xilinx\Vivado\2023.1\data\verilog\src/unisims-y D:\Xilinx\Vivado\2023.1\data\verilog\src/unimacro-y D:\Xilinx\Vivado\2023.1\data\verilog\src/unifast-y D:\Xilinx\Vivado\2023.1\data\verilog\src/retarget
```

LSTM-Kirigaya, 2023-08-11:
我突然发现好像是后续的`-y`前面少了一个空格，那么如果是下面这样，是否可以呢？

```
iverilog -g2012 -o "d:/Project/VSCode/verilog/usart/prj/simulation/icarus/out.vvp" -s des_sample "d:/Project/VSCode/verilog/usart/user/src/des_sample.v"   D:\Xilinx\Vivado\2023.1\data\verilog\src/glbl.v -y D:\Xilinx\Vivado\2023.1\data\verilog\src/xeclib -y D:\Xilinx\Vivado\2023.1\data\verilog\src/unisims -y D:\Xilinx\Vivado\2023.1\data\verilog\src/unimacro -y D:\Xilinx\Vivado\2023.1\data\verilog\src/unifast -y D:\Xilinx\Vivado\2023.1\data\verilog\src/retarget
```

DreamLand7707, 2023-08-12:
> 我突然发现好像是后续的`-y`前面少了一个空格，那么如果是下面这样，是否可以呢？
>
> ```
> iverilog -g2012 -o "d:/Project/VSCode/verilog/usart/prj/simulation/icarus/out.vvp" -s des_sample [...]

情况是一样的
[image]
仍然需要提前到前面

closed 2023-08-15 by LSTM-Kirigaya

DreamLand7707, 2023-08-15:
恢复正常了，正常输出如图：
[image]