Digital-EDA/Digital-IDE #12: [0.3.0 beta] 高亮颜色错误

DuBirdFly, 2023-08-26:
端口的最后一条-高亮颜色错误:
- 最后一条端口且该端口没有被"wire""reg""[xx:xx]""注释"等修饰时, 颜色错误
- 或说最后一条端口只由端口类型与端口名称构成时, 颜色错误

[image]

示例代码:

```verilog
module test_2(
    input            clk,
    // reg/wire/[xx:xx]/注释  至少有一个就不会发生颜色错误
    output reg [7:0] dout  // data out
);
endmodule
module test_3(
    input           clk,
    output          dout
);
endmodule
module test_4(
    output wire     dout,
    input           clk
);
endmodule
module test_5(
    output wire     dout,
    inout           clk
);
endmodule
```

DuBirdFly, 2023-08-26:
GitHub也有这个毛病......乐了
[image]

LSTM-Kirigaya, 2023-12-12:
我们重写了开源的 verilog textmate 文件，这个文件应该解决了：
[image]

closed 2023-12-12 by LSTM-Kirigaya