Digital-EDA/Digital-IDE #29: [0.3.2]param语法错误会弹右下角报错弹窗, TreeView刷新按钮无效

DuBirdFly, 2023-12-12:
当我把模块内的 "parameter START_VALUE = 8'b0000_1111;" 移到模块定义'#'之后时. 发生了弹窗报错如下
(当然我知道不应该有这个分号)

[image]

删除分号后点 "TreeView刷新按钮无效", 必须重启才能恢复

[image]

也就是说我只要不小心在"#"后写parameter写错多写/复制了个分号的话, 我只能重启Vscode才能解决这个问题, 非常折磨

DuBirdFly, 2023-12-12:
测试代码
```verilog
module counter #(
    parameter START_VALUE = 8'b0000_1111;
)(
    input               clk,
    input               rst_n,
    output reg  [7:0]   cnt
);

parameter STEP = 1'b1;

always @(posedge clk or negedge rst_n) begin
    if (!rst_n)
        cnt <= START_VALUE;
    else
        cnt <= cnt + STEP;
end

endmodule

```

DreamLand7707, 2023-12-18:
> 当我把模块内的 "parameter START_VALUE = 8'b0000_1111;" 移到模块定义'#'之后时. 发生了弹窗报错如下 (当然我知道不应该有这个分号) [...]

同样，这个问题很恼火

Nitcloud, 2024-05-20:
我们已更新了新的版本0.3.3，如果还遇到同样的问题请在此留言，一周之后我们将关闭此问题，并视为您的问题已被解决。如果已被关闭且未解决，您可以重新打开此问题。

closed 2025-01-11 by Nitcloud