Digital-EDA/Digital-IDE #34: 文档中的params和ports数反了

jxzsxsp, 2023-12-18:
[image]

DreamLand7707, 2023-12-18:
> ![微信截图_20231218111038](https://private-user-images.githubusercontent.com/3166271/291140726-cba47a2d-dd99-49e5-b302-733e266d0c44.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDI5MDAzOTgsIm5iZiI6MTcwMjkwMDA5OCwicGF0aCI6Ii8zMTY2MjcxLzI5MTE0MDcyNi1jYmE0N2EyZC1kZDk5LTQ5ZTUtYjMwMi03MzNlMjY2ZDBjNDQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQUlXTkpZQVg0Q1NWRUg1M0ElMkYyMDIzMTIxOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzEyMThUMTE0ODE4WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MjNlNGU4MWM4OWY5MjA2NmI5OWJhYTFlODBlNDIyNjJlOTRlMGY3YTQ5Y2E5NzRmMWMzYmIwMzQwNGRkMzg2NCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.-cDpHqCxT3QF-BUAsnd7YMh6Boj6bNZYeFCSlDIK9L8)

我也发现这个问题了

LSTM-Kirigaya, 2023-12-18:
[image]

艹，我的问题

Nitcloud, 2024-05-20:
我们已更新了新的版本0.3.3，如果还遇到同样的问题请在此留言，一周之后我们将关闭此问题，并视为您的问题已被解决。如果已被关闭且未解决，您可以重新打开此问题。

closed 2024-06-13 by Nitcloud