Digital-EDA/Digital-IDE #36: [0.3.2] 代码补全有多个内容完全相同的选项

DreamLand7707, 2023-12-18:
如图所示，输入cl会出现多了相同的clk,影响代码补全的正常使用。
[image]
[image]

DreamLand7707, 2023-12-18:
摸索出来一个规律，这个东西和always块有关系：

[image]

我在1处写下next_ , ctrl + space:
[image]
一切正常

删除
再次在1处写下next_, ctrl + space
[image]
正常

删除
在2处操作：
[image]
出现了重复解析，两次

删除，再次在2处操作
[image]
仍然重复解析，两次，情况没变

删除，在1处操作
[image]
仍然有重复解析，两次

删除，在3处操作：
[image]
重复解析，三次，加了一次

删除，在2处操作：
[image]
重复解析，三次

删除，在1处操作：
[image]
重复解析，三次

删除，在2处重新操作
[image]
重复解析，四次

和always块，解析顺序，之前写过的位置和当前写入的位置都有关系。

LSTM-Kirigaya, 2023-12-18:
感谢您事无巨细的描述，我们会尽快修改并附上 release

commit 801912e7 (2023-12-18): fix #36
closed 2023-12-18 by LSTM-Kirigaya