Digital-EDA/Digital-IDE #45: 重复仿真时报错

firerock1, 2023-12-29:
（在此项目重新提交一下）

- 环境：MacOS Ventura13.6.1，digital IDE 0.3.2版本，iverilog使用homebrew下载
- 目的：对单个模块建立testbench进行仿真
- 报错信息：error: Unknown module type
- 复现过程：
    - 首次打开vscode，点击testbench文件进行仿真，成功，可查看波形图
    - 对testbench进行任意修改（如将分号删去后再加上等）再保存，再次点击仿真，则显示error: Unknown module type，但是模块名称并未进行任何修改

[image]

- 另：在Windows10上重复出现了这个问题，同样最开始会报错
Errors happen when parsing d:/Code/Verilog/exercise/count/user/sim/testbench.v. Error: "RuntimeError: null function or function signature mismatch". Just propose a valuable issue in our github repo 😊

RockRockWhite, 2024-03-28:
The same bug in MACOS

Nitcloud, 2024-05-20:
我们已更新了新的版本0.3.3，如果还遇到同样的问题请在此留言，一周之后我们将关闭此问题，并视为您的问题已被解决。如果已被关闭且未解决，您可以重新打开此问题。

closed 2024-06-13 by Nitcloud