Digital-EDA/Digital-IDE #46: .v源文件未被正确识别

bin0612, 2024-01-05:
[image]
[image]
生成了一个aurora的example design，通过digital ide来查看，发现aurora_64b66b_0_exdes未能正确识别

cncherisher, 2024-01-10:
同样遇到此问题
附上我的verilog源文件
```
`timescale 1ns/1ns

moudle counter_module#(
    parameter                       P_CNT_WIDTH = 8
)(
    input                           i_clk,
    input                           i_rst,
    input                           i_en,
    output [P_CNT_WIDTH - 1 : 0]    o_cnt 
);

endmodule
```
Errors happen when parsing d:/FPGA/bro_qi/class1_project/Prj_04/user/src/counter_module.v. Error: "RuntimeError: null function or function signature mismatch". Just propose a valuable issue in our github repo 😊

Rlxzmdd, 2024-01-30:
请问解决这个问题了么？我每次刚启动都可以正常识别，第二次就不行了

Nitcloud, 2024-05-20:
我们已更新了新的版本0.3.3，如果还遇到同样的问题请在此留言，一周之后我们将关闭此问题，并视为您的问题已被解决。如果已被关闭且未解决，您可以重新打开此问题。

closed 2024-06-13 by Nitcloud