Digital-EDA/Digital-IDE #63: 【报错】RuntimeError: null function or function signature mismatch

mathsmatics, 2024-04-29:
[image]
复制源代码进文件夹之后，然后把文件夹拖到vscode，然后显示是否生成json文件，点的是，然后点文件的时候就会报这个错，然后报错的文件，就无法进行变量关联

mathsmatics, 2024-04-29:
就是写代码的时候，定义过的变量不能自动补全，感觉问题出现的原因和这个报错有关

mathsmatics, 2024-04-29:
我尝试所有文件新建文件，再粘贴内容的，粘贴下面这段的时候，报错：
Errors happen when parsing d:/projectFile/ModelSim/OpenMIPS/rtl/ex_mem.v. Error: "RuntimeError: null function or function signature mismatch". Just propose a valuable issue in our github repo 😊
```verilog
//////////////////////////////////////////////////////////////////////
// Description: EX/MEM阶段的寄存器
//////////////////////////////////////////////////////////////////////

`include "defines.v"

module ex_mem(
	input wire						clk,
	input wire						rst,

	//来自控制模块的信息
	input wire[5:0]					stall,

	//来自执行阶段的信息	
	input wire[`RegAddrBus]			ex_wd,
	input wire						ex_wreg,
	input wire[`RegBus]				ex_wdata,
	input wire[`RegBus]				ex_hi,
	input wire[`RegBus]				ex_lo,
	input wire						ex_whilo,
	
	input wire[`DoubleRegBus]		hilo_i,
	input wire[1:0]					cnt_i,

	//送到访存阶段的信息
	output reg[`RegAddrBus]			mem_wd,
	output reg						mem_wreg,
	output reg[`RegBus]				mem_wdata,
	output reg[`RegBus]				mem_hi,
	output reg[`RegBus]				mem_lo,
	output reg						mem_whilo,

	output reg[`DoubleRegBus]		hilo_o,
	output reg[1:0]					cnt_o
);

	always @ (posedge clk) begin
		if(rst == `RstEnable) begin
			mem_wd <= `NOPRegAddr;
			mem_wreg <= `WriteDisable;
			mem_wdata <= `ZeroWord;
			mem_hi <= `ZeroWord;
			mem_lo <= `ZeroWord;
			mem_whilo <= `WriteDisable;
			hilo_o <= {`ZeroWord,`ZeroWord};
			cnt_o <= 2'b0;
		end 
		else if(stall[3] == `Stop && stall[4] == `NoStop)begin
			mem_wd <= `NOPRegAddr;
			mem_wreg <= `WriteDisable;
			mem_wdata <= `ZeroWord;
			mem_hi <= `ZeroWord;
			mem_lo <= `ZeroWord;
			mem_whilo <= `WriteDisable;
			hilo_o <= hilo_i;
			cnt_o <= cnt_i;
		end
		else if(stall[3] == `NoStop)begin
			mem_wd <= ex_wd;
			mem_wreg <= ex_wreg;
			mem_wdata <= ex_wdata;
			mem_hi <= ex_hi;
			mem_lo <= ex_lo;
			mem_whilo <= ex_whilo;
			hilo_o <= {`ZeroWord,`ZeroWord};
			cnt_o <= 2'b0;
		end
		else begin
			hilo_o <= hilo_i;
			cnt_o <= cnt_i;
		end //if
	end //always
			

endmodule
```

reoLantern, 2024-05-11:
+1，遇到了同样的问题，导致大纲、定义跳转也无法使用了

Jinghao-Tu, 2024-06-11:
0.3.2 正常工作, 0.3.3 不行.

DanielCheung16, 2024-12-24:
遇到了同样的问题。在你现在这个rtl目录下只留下.v文件就可以了。其它都是Digital IDE自己创建的。删除了就正常了

closed 2025-01-11 by Nitcloud