Digital-EDA/Digital-IDE #66: [0.3.3 beta] 含参数模型的例化存在问题以及拓展快捷键失效的问题

light-ly, 2024-05-13:
在尝试 0.3.3-beta 之后，发现依然自动例化带参数的模块。同时在 treeview 中，修改带参数的模块实例化为正确的格式，没有被插件正确识别，如图
1. 错误的自动例化格式
[image]
2. 格式修改正确后不能被 treeview 正确识别
[image]
3. 同时还存在一个小问题，即使用 `Alt+I` 快捷键时，vscode 报错，使用 `Ctrl+Shift+p`  寻找对应的命令并执行时没有此问题
[image]

辛苦各位大佬修复一下

light-ly, 2024-05-13:
问题继承自 #64

light-ly, 2024-05-13:
[rtl.zip](https://github.com/Digital-EDA/Digital-IDE/files/15290143/rtl.zip)
链接是工程中的 rtl 文件

LSTM-Kirigaya, 2024-05-13:
非常详细的复现代码，感谢您提供的数据，我们尽快解决。

LSTM-Kirigaya, 2024-05-13:
我认为是你的 property.json 没有设置好，当你把 property.json 设置为如下的：

```json
{
	"toolChain": "xilinx",
	"prjName": {
		"PL": "template"
	},
	"soc": {
		"core": "none"
	},
	"enableShowLog": false,
	"device": "none",
    "arch": {
        "hardware": {
            "src": "./rtl"
        }
    }
}
```

[image]

light-ly, 2024-05-14:
我找到我的问题了，麻烦大佬了，实际上问题是由于我的代码不规范造成的。
在 RegFile.v 和 Ram.v 两个文件中，我在声明参数时没有添加 parameter 关键字，因为能通过 verilator 的编译，所以我忽视了这一问题，但这应该是使得插件的匹配模式失效了，所以没有承认我这两个模块，也导致了自动实例化时没能正确生成格式，不好意思，由于我的疏忽耽误了各位的时间。

light-ly, 2024-05-14:
修改该问题后插件各个功能已经恢复正常，0.3.3 beta 已经能够正确自动例化代码，感谢佬的协助

closed 2024-05-14 by light-ly