Digital-EDA/Digital-IDE #70: 采用iverilog生成的VCD貌似无法解析仿真数据

chesswei1, 2024-05-23:
采用iverilog生成的VCD貌似无法解析仿真数据
iverilog -o testbench crc8_sgpio_hisport.v dual_clk_fifo.v  i2c_over_hisport_tb.v i2c_over_hisport.v hisport_rx.v hisport_rx_buf.v hisport_top_tx_rx.v hisport_tx.v hisport_tx_buf.v i2c_command_arbit.v i2c_consequences_dispatch.v i2c_ctrl_fifo.v i2c_master.v lbus_rx.v lbus_rx_dispatch.v lbus_tx.v lbus_tx_dispatch.v macro.v
vvp -n testbench -lxt2

LSTM-Kirigaya, 2024-06-05:
你可以看一下生成的 vcd 文件用文本编辑器打开是不是乱码，如果是乱码，说明这个vcd属于 fstdb 协议下的。0.3.3 版本的 Digital IDE 暂时不支持 fstdb 协议的 vcd，我们会在下个版本开始支持。

Nitcloud, 2024-11-04:
对于二进制的vcd文件的解析暂不支持，请给出你的iverilog的版本。

YouXiaoquan, 2025-02-05:
iverilog V12报 -I参数错误。
iverilog V11生成的VCD波形无法打开，一直转圈圈。但是用WaveTrace打开该vcd文件正常。

LSTM-Kirigaya, 2025-02-05:
> iverilog V12报 -I参数错误。 iverilog V11生成的VCD波形无法打开，一直转圈圈。但是用WaveTrace打开该vcd文件正常。

方便向我们提供 vcd 文件或是生成 vcd 的项目代码吗？

YouXiaoquan, 2025-02-05:
附件中是该vcd文件，请查收

 

另外，我常用的装了很多插件的vs code，可以用Digital-EDA观看该波形文件。而我用全新的虚拟机win10或win11 + vs code  + Digital-EDA，就无法解析该仿真数据了，只会在加载的时候转圈圈了。

YouXiaoquan, 2025-02-06:
目前问题最新复现细节：
我利用Win to Go技术，在移动硬盘放了两个系统镜像，分别是：Win11 + VS Code + Digital IDE(0.4.0)和Win10 + VS Code + Digital IDE(0.4.0)，两个系统中均存放相同的VCD文件（[vcdfile.zip](https://github.com/user-attachments/files/18686282/vcdfile.zip）。在较新的电脑（七代i7）上，上述两个系统中的VCD文件均能够渲染；在旧电脑（二代i7）上，上述两个统统中的VCD文件均不能渲染。

LSTM-Kirigaya, 2025-02-07:
这个复现有点麻烦，因为我们常规的开发和测试环境都不包含虚拟机环境。虚拟机软件本身对于渲染相关的支持确实可能如您所言，不支持最新的特性。不过还是感谢你的贡献。

我在 x86 windows 上是可以完成vcd的解析和渲染的：

[image]

如果急着使用的话，可以先尝试别的 vcd 渲染器，比如 surfer，虚拟机的环境我们这几个月会尽快去支持的。

LSTM-Kirigaya, 2025-02-07:
渲染的时候出现的错误方便截图发给我们吗？具体做法：
1. 打开 vscode，开启一个 dide 的项目
2. 按下 F1 输入 toggle，找到 Developer: Toggle Developer Tools，按下打开控制台，找到控制台输出信息
3. 截图发给我们

[image]

YouXiaoquan, 2025-02-07:
> 这个复现有点麻烦，因为我们常规的开发和测试环境都不包含虚拟机环境。虚拟机软件本身对于渲染相关的支持确实可能如您所言，不支持最新的特性。不过还是感谢你的贡献。
> 
> 我在 x86 windows 上是可以完成vcd的解析和渲染的：
> 
> [image]
> 
> 如果急着使用的话，可以先尝试别的 vcd 渲染器，比如 surfer，虚拟机的环境我们这几个月会尽快去支持的。

只要CPU是四代i3以上的机器，无论是实体机还是虚拟机都可以正常渲染。只要CPU是三代及以下的都不能正常渲染。

YouXiaoquan, 2025-02-07:
> 渲染的时候出现的错误方便截图发给我们吗？具体做法：
> 
> 1. 打开 vscode，开启一个 dide 的项目
> 2. 按下 F1 输入 toggle，找到 Developer: Toggle Developer Tools，按下打开控制台，找到控制台输出信息
> 3. 截图发给我们
> 
> [image]

截图信息如下：

[image]