Digital-EDA/Digital-IDE #76: 悬停提示识别到错误内容

narutozxp, 2024-06-28:
悬停提示，会显示标识对应的注释内容,其识别机制如下：
>其中标识对应的注释的内容为
>标识被定义的所在行后的行注释
>标识被定义之前的行注释以及块注释(遇到非注释部分即停止)

但是这种识别方案会导致将文件顶部的注释（用于表示文件的模块版本之类的信息）给识别到了：
[image]
因此建议将识别机制修改为：
>标识被定义的所在行后的行内注释（同一行）
>标识被定义之前的行注释以及块注释(需要紧贴着定义的行)

除此之外，生成的schematic中，没有显示常数（在本例中是常数0）
[image]

12113004, 2024-08-23:
[image]
没错，悬停显示有问题

12113004, 2024-08-23:
我回退到之前0.3.2版本后显示正常

Nitcloud, 2024-09-06:
悬停提示将会在0.4.0版本解决，生成的schematic问题将会在0.4.2版本中解决。

closed 2025-01-11 by Nitcloud