Digital-EDA/Digital-IDE #85: 自动生成文档功能bug

GarenZZ, 2024-10-10:
在同一个目录下，jtag_para.v可以生成文档，而para_st.v不能生成文档，错误报告显示para_st.v不处于parse list目录。此时使用teroshdl可以正常生成文档
[image]
[image]
[image]
[image]
[image]

GarenZZ, 2024-10-10:
在解析目录中，大约有一半的源文件无法使用digital-ide生成文档

Nitcloud, 2024-11-04:
请给出插件的版本号，并使用0.4.0重新测试一下，新版本应该是解决该问题，若无后续更新则将在一个月后关闭此issue。

closed 2025-01-11 by Nitcloud