Digital-EDA/Digital-IDE #105: [0.4.0]testbench中右键快速仿真提示“... is is not a hdlFile", 原工程直接切换回0.3.3版本可以正常仿真。

alwaysmy, 2024-11-12:
vscode相关环境：
这里配置了两个profile，一个用来安装0.3.3的版本，一个用来安装0.4.0的版本，后面切换插件测试是通过切换vscode profile来实现的，可能是这个会有影响（因为我发现从0.4.0的profile切换到0.3.3之后，右键菜单会变成英文，切换回来0.4.0之后右键菜单还是英文）。

复现步骤：

- 1、新建工程文件夹
- 2、右键code打开
- 3、点击DigitalIDE插件图标，等右下角弹出配置property.json的提示后点确定
- 4、回到vscode资源管理器文件目录下，在src下新建模块源文件，编写源文件（这里复制了之前验证过的内容）
- 5、在sim下新建testbench源文件 ( 分别试过手工新建文件后编写以及使用右键模块创建testbench模板 ，不影响结果）
- 6、在testbench源文件中右键对当前文件仿真
- 7、这时候会提示当前文件property.json
- 8、通过切换vscode profile, 回到0.3.3版本的插件，依旧testbench源文件中右键对当前文件仿真，可以正常仿真。

其他测试：
之前用0.3.3创建的工程切换到0.4会出现同样错误，回到0.3.3下正常，**直接卸载0.3.3版本插件后。0.4.0的插件正常**。

如果是vscode本身的原因的话，也许文档里面后面说一下不要装两个版本？

Nitcloud, 2024-11-13:
工程建好之后重启一下vscode后是否能正常仿真？

alwaysmy, 2024-11-13:
不能，重启电脑也不行（实际上之前反复安装插件重启过vscode很多次）；另外刚刚试了一下在个新的电脑上安装vscode后安装插件后仿真这个工程依然能复现这个问题，这里附上我用来测试的工程文件。
[TestDesign.zip](https://github.com/user-attachments/files/17725720/TestDesign.zip)

closed 2025-01-11 by Nitcloud