Digital-EDA/Digital-IDE #108: [0.4.0] vcd文件打不开，一直转圈“加载中"

alwaysmy, 2024-11-16:
步骤：
新建文件夹，右键 code 打开
弹出面板确认新建 property.json
在文件树下src内新建 .v 源文件，sim下新建testbenche.v源文件
仿真后生成wave.vcd.
[image]

尝试切换WaveTrace插件可以正常打开这个wave.vcd。
注：0.3.3也打不开，任意工程都不行

ylingu, 2024-11-18:
遇到了同样的问题，以文本方式打开并非 #70  中提到的乱码，vcd文件内容如下：
```
$date
	Fri Nov 08 01:26:24 2024
$end
$version
	Icarus Verilog
$end
$timescale
	1ns
$end
$scope module testbench $end
$var wire 1 ! clk_div $end
$var reg 1 " clk_sys $end
$scope module uut $end
$var wire 1 " clk $end
$var wire 1 # reset $end
$var parameter 32 $ DIVISOR $end
$var reg 1 ! clk_div $end
$var reg 32 % counter [31:0] $end
$var integer 32 & divisor [31:0] $end
$upscope $end
$upscope $end
$enddefinitions $end
$comment Show the parameter values. $end
$dumpall
b10 $
$end
#0
$dumpvars
b1 &
b0 %
0#
0"
0!
$end
#1
1!
1"
#2
0"
#3
0!
1"
#4
0"
#5
1!
1"
#6
0"
#7
0!
1"
#8
0"
#9
1!
1"
#10
0"
#11
0!
1"
#12
0"
#13
1!
1"
#14
0"
#15
0!
1"
#16
0"
#17
1!
1"
#18
0"
#19
0!
1"
#20
0"
#21
1!
1"
#22
0"
#23
0!
1"
#24
0"
#25
1!
1"
#26
0"
#27
0!
1"
#28
0"
#29
1!
1"
#30
0"
#31
0!
1"
#32
0"
#33
1!
1"
#34
0"
#35
0!
1"
#36
0"
#37
1!
1"
#38
0"
#39
0!
1"
#40
0"
#41
1!
1"
#42
0"
#43
0!
1"
#44
0"
#45
1!
1"
#46
0"
#47
0!
1"
#48
0"
#49
1!
1"
#50
0"
#51
0!
1"
#52
0"
#53
1!
1"
#54
0"
#55
0!
1"
#56
0"
#57
1!
1"
#58
0"
#59
0!
1"
#60
0"
#61
1!
1"
#62
0"
#63
0!
1"
#64
0"
#65
1!
1"
#66
0"
#67
0!
1"
#68
0"
#69
1!
1"
#70
0"
#71
0!
1"
#72
0"
#73
1!
1"
#74
0"
#75
0!
1"
#76
0"
#77
1!
1"
#78
0"
#79
0!
1"
#80
0"
#81
1!
1"
#82
0"
#83
0!
1"
#84
0"
#85
1!
1"
#86
0"
#87
0!
1"
#88
0"
#89
1!
1"
#90
0"
#91
0!
1"
#92
0"
#93
1!
1"
#94
0"
#95
0!
1"
#96
0"
#97
1!
1"
#98
0"
#99
0!
1"
#100
0"
```

closed 2025-01-11 by Nitcloud

YouXiaoquan, 2025-02-05:
我也遇到同样的问题，环境为win10 + VS code + iverilog 11 + Digitai + Digital EDA 0.4.0;
但是偶然发现另一台win11电脑就可以正常显示波形